espressif/qemu #137: ESP32S3 PSRAM not detected if octal type or greater than 8MB

ysard, 2025-08-25:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

Hi, 
I have tested numerous combinations of frequency, size, and memory type for the ESP32S3 under ESP-IDF 5.4.1. Some combinations do not work.

QEMU version (--version): QEMU emulator version 9.2.2 (esp_develop_9.2.2_20250817)

Tested cases:
```
80MHz Quad
ok  2MB
ok  4MB
ok  8MB
nok 16MB
nok 32MB

40MHz Quad
ok 4MB
ok 8MB
nok 16MB
nok 32MB

40MHz Octal
nok 4MB
nok 8MB
nok 16MB
nok 32MB

80MHz Octal
nok 4MB
nok 8MB
nok 16MB
nok 32MB
```

### Actual behavior (suspected bug)

The examples of console output below show that there seem to be two types of errors (PSRAM ID for Octal memory, `<cached disabled>` for Quad memory).

### Error logs or terminal output

```shell
## 40MHz, Octal, 16MB

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x4 (SPI_FLASH_BOOT)
SPIWP:0xee
mode:QIO, clock div:2
load:0x3fce2810,len:0x15a0
load:0x403c8700,len:0x4
load:0x403c8704,len:0xd20
load:0x403cb700,len:0x2f00
entry 0x403c8920
I (36) boot: ESP-IDF 5.4.1 2nd stage bootloader
I (39) boot: compile time Aug 25 2025 19:50:13
I (40) boot: Multicore bootloader
I (46) boot: chip revision: v0.0
I (47) boot: efuse block revision: v1.0
I (50) boot.esp32s3: Boot SPI Speed : 40MHz
[... 209 lines cut ...]
    -global driver=nvram.esp32s3.efuse,property=drive,value=efuse \
    -serial file:output.log \
    -d guest_errors \
    -m 32M
```

### Project release version

9.2.2 (esp_develop_9.2.2_20250817)

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Linux

### Operating system version

Debian Testing

### Shell

ZSH

ysard, 2025-08-26:
@o-marshmallow Hi, thank you for your quick response, It now works for Quad SPI (all sizes). Still nothing for Octal memory.
It is not a big problem but if would be great if a note were added to the documentation (idf 5.4 limitation + quad spi support only) ?

o-marshmallow, 2025-08-28:
@ysard Using the latest release of QEMU (`esp-develop-9.2.2-20250817`), and ESP-IDF v5.4.1, both 16MB and 32MB Octal PSRAM work, using the following commands:
```
$ qemu-system-xtensa -nographic -machine esp32s3 -drive file=<path_to_img>,if=mtd,format=raw -m 32M -global driver=ssi_psram,property=is_octal,value=true

I (54) boot: ESP-IDF v5.4.1 2nd stage bootloader
...
I (73) boot.esp32s3: Boot SPI Speed : 80MHz
I (74) boot.esp32s3: SPI Mode       : SLOW READ
...
I (229) octal_psram: vendor id    : 0x0d (AP)
I (229) octal_psram: dev id       : 0x02 (generation 3)
I (230) octal_psram: density      : 0x07 (256 Mbit)
I (231) octal_psram: good-die     : 0x01 (Pass)
I (231) octal_psram: Latency      : 0x01 (Fixed)
I (231) octal_psram: VCC          : 0x00 (1.8V)
I (232) octal_psram: SRF          : 0x01 (Fast Refresh)
I (232) octal_psram: BurstType    : 0x01 (Hybrid Wrap)
I (233) octal_psram: BurstLen     : 0x01 (32 Byte)
I (233) octal_psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (234) octal_psram: DriveStrength: 0x00 (1/1)
I (238) esp_psram: Found 32MB PSRAM device
I (239) esp_psram: Speed: 40MHz
```
and:
```
$ qemu-system-xtensa -nographic -machine esp32s3 -drive file=<path_to_img>,if=mtd,format=raw -m 16M -global driver=ssi_psram,property=is_octal,value=true

I (60) boot: ESP-IDF v5.4.1 2nd stage bootloader
...
I (79) boot.esp32s3: Boot SPI Speed : 80MHz
I (80) boot.esp32s3: SPI Mode       : SLOW READ
...
I (248) octal_psram: vendor id    : 0x0d (AP)
I (249) octal_psram: dev id       : 0x02 (generation 3)
I (250) octal_psram: density      : 0x05 (128 Mbit)
I (250) octal_psram: good-die     : 0x01 (Pass)
I (251) octal_psram: Latency      : 0x01 (Fixed)
I (251) octal_psram: VCC          : 0x00 (1.8V)
I (252) octal_psram: SRF          : 0x01 (Fast Refresh)
I (252) octal_psram: BurstType    : 0x01 (Hybrid Wrap)
I (252) octal_psram: BurstLen     : 0x01 (32 Byte)
I (253) octal_psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (253) octal_psram: DriveStrength: 0x00 (1/1)
I (257) esp_psram: Found 16MB PSRAM device
I (258) esp_psram: Speed: 40MHz
```

Make sure you passed the parameter `-global driver=ssi_psram,property=is_octal,value=true` to QEMU to tell the PSRAM to act like an octal PSRAM

ysard, 2025-08-28:
Yes indeed, it works fine from 4MB to 32MB, @80MHz and @40MHz, many thanks!

I don't think I could have created the command on my own; this command deserves to be included in the documentation. I think you can close the issue now ;)

o-marshmallow, 2025-08-29:
@ysard My bad, I thought it was, I am going to add it to the documentation 👍

o-marshmallow, 2025-08-29:
@ysard Added: https://github.com/espressif/esp-toolchain-docs/blob/main/qemu/esp32s3/README.md#enabling-octal-psram